techyian/MMALSharp #12: Colour conversions

techyian, 2017-04-07:
Provide support for converting between different colour spaces. This is for use with the `AnnotateImage` method in `MMALEncoderComponent`.

daniel-lerch, 2018-05-05:
Hi Ian,

in 7ba945d09e04af2d36a8086ff5215dc7b9bd9ee9 you have added extension methods for byte <-> float conversion.
This method absolutely makes sense:
```csharp
public static byte ToByte(this float val);
```
But this method's name is irritating:
```csharp
public static float FromByte(this byte val);
```
I would either call `float FromByte(byte)` on a static class, not as an extension method or rename the method to `float ToFloat(byte)` or `float ToSingle(byte)`. Microsoft's APIs use `Single` for method names and `float` only as C# keyword.

All other changes I've seen so far on this branch look very good 👍 

Daniel

techyian, 2018-05-05:
Thanks Daniel, it's all been a bit of a brain dump in this branch so far and is very experimental to say the least! 😀 I'll get that method renamed shortly. 

Ian

closed 2018-05-16 by techyian