techyian/MMALSharp #31: MMALNoMemoryException

fabien-chevalley, 2017-12-29:
Hi,

I got the following exception during instanciation of MMALCamera : 
```
An exception of type 'MMALSharp.MMALNoMemoryException' occurred in System.Private.CoreLib.dll but was not handled in user code: 'Out of memory. Unable to create component'
   at MMALSharp.MMALCallerHelper.MMALCheck(MMAL_STATUS_T status, String prefix)
   at MMALSharp.MMALComponentBase.CreateComponent(String name)
   at MMALSharp.MMALComponentBase..ctor(String name)
   at MMALSharp.Components.MMALCameraComponent..ctor()
   at MMALSharp.MMALCamera..ctor()
   at MMALSharp.MMALCamera.<>c.<.cctor>b__38_0()
   at System.Lazy`1.ViaFactory(LazyThreadSafetyMode mode)
   at System.Lazy`1.ExecutionAndPublication(LazyHelper executionAndPublication, Boolean useDefaultConstructor)
   at System.Lazy`1.CreateValue()
   at MMALSharp.MMALCamera.get_Instance()

Do you have any idea of my mistake ?

For informations, I'm running the code on raspbian9 with .netcore2.1

Fabien

techyian, 2017-12-30:
Hi, could you post a sample of code you're running please so I can try locally? Also, what is the memory split for the GPU?

Thanks,
Ian

fabien-chevalley, 2017-12-30:
ok, I foud my mistake... I didn't activate the camera module on raspi-config !

Now, I can correctly grab image from raspistill but MMALSharp return 0byte image. I'm going to dig a little bit to try to find my mistake

fabien-chevalley, 2017-12-30:
it's an issue of the remote debugger, if I run my code without vsbdg everthing wokrs.
Thank's for helping me and for the library !

closed 2017-12-30 by fabien-chevalley