techyian/MMALSharp #35: Include XML Documentation in NuGet Package

daniel-lerch, 2018-04-02:
Visual Studio allows to make Documentation in Code:
```cs
/// <summary>
/// This class provides an interface to the Raspberry Pi camera module. 
/// </summary>
public sealed class MMALCamera 
{
}
```
But this documentation is not shown as long as you don't generate an XML file at build and include this file in your NuGet package. In Visual Studio you can go to `Project Settings->Build` and enable `XML documentation file`. This will generate a `MMALSharp.xml` in your output directory which you can include to your NuGet package. After this all developers that are using your library will see your documentation in Visual Studio.

techyian, 2018-04-04:
Hi Daniel,

Thanks for bringing this to my attention :) I'll get it fixed ASAP.

Ian

daniel-lerch, 2018-04-04:
Hi Ian,

I have fixed this issue locally as I made an own build.
Do you would like me to make a pull request?

Daniel

techyian, 2018-04-04:
Sure, I'll check it over and merge in.

Thanks again,
Ian.

PR #36 (2018-04-04): Configured XML documentation output
I configured the output of XML documentation files. The compiler will now create an XML file in the output directory which resolves #35 .

The `<Service Include="" />` tag in ` tests/MMALSharp.Tests/MMALSharp.Tests.csproj` was automatically made by Visual Studio in order to optimize startup performance with XUnit ([see stackoverflow](https://stackoverflow.com/questions/18614342/what-is-service-include-in-a-csproj-file-for)).
closed 2018-04-04 by techyian