techyian/MMALSharp #37: Update XML documentation

techyian, 2018-04-04:
Update XML documentation for currently implemented Managed areas of MMALSharp.

Suppress warnings in native areas.

daniel-lerch, 2018-04-05:
Hi Ian,

I have started to add XML documentation to the code. Because I first came in touch with MMAL five days ago I don't understand all your code yet but I'll give my best to contribute to the documentation.

How requently do you want to get pull requests?

Daniel

techyian, 2018-04-05:
Hi Daniel,

Thank you for getting involved in the project. I would encourage you to look into the Native MMAL API that the Raspberry Pi team maintain first if you're new to it all. 

The native API docs can be found [here](http://www.jvcref.com/files/PI/documentation/html/) and help with understanding what each of the native calls do.

For more theory into the actual workings of MMAL and the Raspberry Pi camera itself, the PiCamera python library has extensive documentation [here](http://picamera.readthedocs.io/en/latest/). I'd certainly like this project to have that kind of documentation in the future, but as I'm sure you can appreciate, it all takes up a lot of time.

There's quite a lot to the API itself, but it will give you a good footing to be able to contribute to MMALSharp.

Ian.

closed 2018-11-27 by techyian