techyian/MMALSharp #39: "Type could not be marshaled" Error

ianByrne, 2018-04-05:
I'm getting this error on a Raspberry Pi 3B+ running Raspbian:

> Type could not be marshaled because the length of an embedded array instance does not match the declared length in the layout.

My .Net Core app is pretty much [the example from the Wiki](https://github.com/techyian/MMALSharp/wiki/Examples-0.3) (although it seems the `MMALEncoding` values are different?):

```csharp
MMALCamera cam = MMALCamera.Instance;

AsyncContext.Run(async () =>
{
    using (var imgCaptureHandler = new ImageStreamCaptureHandler("/home/pi/images/", "jpg"))
    {
        await cam.TakePicture(imgCaptureHandler, MMALEncoding.MMAL_ENCODING_JPEG, MMALEncoding.MMAL_ENCODING_I420);
    }
});

cam.Cleanup();
```

Stack trace:

> at System.Runtime.InteropServices.Marshal.StructureToPtr(Object structure, IntPtr ptr, Boolean fDeleteOld)
> at MMALSharp.Components.MMALImageEncoder.AddExifTag(ExifTag exifTag)
> at MMALSharp.Components.MMALImageEncoder.<AddExifTags>b__24_0(ExifTag c)
> at System.Collections.Generic.List`1.ForEach(Action`1 action)
> at MMALSharp.Components.MMALImageEncoder.AddExifTags(ExifTag[] exifTags)
> at MMALSharp.Components.MMALImageEncoder.ConfigureOutputPort(Int32 outputPort, MMALEncoding encodingType, MMALEncoding pixelFormat, Int32 quality, Int32 bitrate)
> at MMALSharp.MMALCamera.<TakePicture>d__25.MoveNext()
> --- End of stack trace from previous location where exception was thrown ---
> at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
> at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
> at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
> at PiCameraProject.Program.<>c__DisplayClass3_0.<<Main>b__0>d.MoveNext() in C:\Projects\PiCameraProject\PiCameraProject\Program.cs:line 42
> --- End of stack trace from previous location where exception was thrown ---
> at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
> at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
> at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
> at Nito.AsyncEx.Synchronous.TaskExtensions.WaitAndUnwrapException(Task task)
> at Nito.AsyncEx.AsyncContext.<>c__DisplayClass15_0.<Run>b__0(Task t)
> at System.Threading.Tasks.ContinuationTaskFromTask.InnerInvoke()
> at System.Threading.Tasks.Task.<>c.<.cctor>b__276_1(Object obj)
> at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
> at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot)
> --- End of stack trace from previous location where exception was thrown ---
> at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
> at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
> at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
> at Nito.AsyncEx.Synchronous.TaskExtensions.WaitAndUnwrapException(Task task)
> at Nito.AsyncEx.AsyncContext.Run(Func`1 action)
> at PiCameraProject.Program.Main(String[] args) in C:\Projects\PiCameraProject\PiCameraProject\Program.cs:line 36

ianByrne, 2018-04-05:
I dug around [myget](https://www.myget.org/gallery/mmalsharp) and saw that there's a new version (128). I switched to this and all is working great now :) The wiki install instructions has an older version.

Many thanks, it's a great tool!

closed 2018-04-05 by ianByrne

techyian, 2018-04-05:
Hi Ian,

Yes, that bug was lying around for a while and only affected .NET Core apps but was fixed for the v0.3 release. 

If you're interested, the issue was happening because when we do the `MMAL_PARAMETER_EXIF_T` marshalling [here](https://github.com/techyian/MMALSharp/blob/dev/src/MMALSharp/Components/EncoderComponents/MMALImageEncoder.cs#L134), .NET Core requires arrays that have a `SizeConst` set against them to be exactly that size. It caught me out for a while because Mono doesn't explicitly require the array being marshalled to be the same size, it can be smaller if needs be and it'll handle it no problem.

Glad you're enjoying it by the way :)