techyian/MMALSharp #45: Safer pointer allocation

techyian, 2018-04-13:
When explicitly allocating pointers, we need to ensure that these are correctly freed in the event of an exception occurring. 

Wrap `MMALCheck` calls in a try/finally and make sure we free any pointers in use.

daniel-lerch, 2018-04-14:
Maybe it would be useful in some classes to additionally override the Finalizer and free pointers. This is much slower than directly freeing them when an exception occurs or in a Dispose call. But I would prefer the pointers to  get freed by the slow garbage collector rather than maybe never because someone forgets a Dispose call.

techyian, 2018-04-14:
Not convinced on Finalizers if I'm honest. All components implement 'IDisposable' and resources are cleaned up in the Dispose methods. It's up to the developer to wrap these in 'using' statements. This issue only applies to a small number of areas where we explicitly call 'Marshal.AllocHGlobal' so I think try/finally is sufficient to resolve.

techyian, 2018-04-24:
Resolved in c9adc78

closed 2018-04-24 by techyian