techyian/MMALSharp #51: `ResizerComponent.ConfigureOutputPort` doesn't work

daniel-lerch, 2018-05-13:
Hi Ian,

I am still trying to get an uncompressed image from MMALSharp.
At first I tried your approach from `TakeRawPicture` but got an EINVAL when I tried to commit the image format ARGB. Then I searched with the new utility method from #17 for available encodings.
Finally I tried many versions with ResizerComponent all resulting in an EINVAL at ConfigureOutputPort. No matter whether I use `I420`, `BGRA` or `YUYV`, it always crashes.

Here is my code:
```csharp
public int Width => Resolution.As8MPixel.Width;
public int Height => Resolution.As8MPixel.Height;

public void Initialize()
{
    MemoryCaptureHandler stillHandler = new MemoryCaptureHandler(null);
    MMALResizerComponent stillResizer = new MMALResizerComponent(Width, Height, stillHandler);
    MMALNullSinkComponent previewSink = new MMALNullSinkComponent();

    MMALCameraConfig.StillResolution = Resolution.As8MPixel;
    cam.ConfigureCameraSettings();

    PrintEncoding(cam.Camera.StillPort.GetSupportedEncodings(), "Still Port");
    PrintEncoding(stillResizer.Outputs[0].GetSupportedEncodings(), "Resizer");

    stillResizer.ConfigureInputPort(MMALEncoding.OPAQUE, MMALEncoding.I420, cam.Camera.StillPort);
    stillResizer.ConfigureOutputPort(MMALEncoding.I420, MMALEncoding.I420, 0);
    cam.Camera.PreviewPort.ConnectTo(previewSink);
    cam.Camera.StillPort.ConnectTo(stillResizer);
}
```
The `MemoryCaptureHandler` simply writes all bytes on a MemoryStream.

Daniel

techyian, 2018-05-13:
Strange. Seems to be working fine for me under Mono 4.6.2. My code is fairly similar to yours except is wrapped in `AsyncContext.Run()`, although that wouldn't cause the EINVAL. Which runtime are you using with your code? Also, which Pi and Camera Module are you using?

daniel-lerch, 2018-05-13:
I am using a Raspberry Pi 3B with the Sony IMX219 camera module.
On Raspbian I am running an [Avalonia](https://github.com/AvaloniaUI/Avalonia) Application which should provide its own async context. But the application crashes before calling any async method.
The application is built on .NET Core 2.0 and published in Debug configuration as a self-contained app for _linux-arm_.

techyian, 2018-05-13:
Ok, thanks. I'll give it a quick test on .NET Core.

daniel-lerch, 2018-05-13:
Hi Ian,

here is requested output:
```
pi@rasperrypi:/opt/vd/bin $ ./vcgencmd version

Mar 23 2018 16:35:52 
Copyright (c) 2012 Broadcom
version d92349b3d22c277e44d508a2aae6af2b1e5bb224 (clean) (release)
```
[Gist for Stacktrace and logfile](https://gist.github.com/daniel-lerch/838ebc139b163b3c1e8c74d9b8860bad)
Note: This version uses `I420` as output encoding from the ResizerComponent. Using other encodings ended up in the same Exception but in some cases twice the output:
`mmal: mmal_vc_port_info_set: failed to set port info (3:0): EINVAL`
`mmal: mmal_vc_port_set_format: mmal_vc_port_info_set failed 0x1bebf10 (EINVAL)`

Thank you very much!

Daniel

techyian, 2018-05-15:
Hi Daniel,

Sorry for the delay in getting back to you - it's been a busy few days. I've updated to the latest firmware and still not having the issues you're experiencing. I have however found a separate issue which is resolved in 4524186. 

At this point it might be worthwhile making a fresh application without the UI framework based on the Resizer example in the Wiki and see if you have any luck.

What would also be helpful is if you can follow the steps [here](https://github.com/techyian/MMALSharp/wiki/Configuration#debugging) regarding `vcdbg` and make a Gist of the output for me - that'll print out a lot of low level debugging info from the firmware to the console and is handy when issues like this crop up.

Thanks,
Ian

daniel-lerch, 2018-05-19:
Hi Ian,

sorry for this delay. Until today I was not able to do more tests.
I have pulled e616b3a4 and built a new Console Application from scratch using my old settings. Again I got an EINVAL. But once I copied your example code from the Wiki it worked. My first mistake was setting the resolution of the ResizerComponent to 0×0 or 65536×65536. However this mistake could not have happened in my lasts attempts with Avalonia where I used `Resolution.As8MPixel`.

Now it works with both projects, Console Application and Avalonia UI. Many issues may have contributed to this error. Avalonia is not stable at all on Raspbian and my power supply had problems when moving the Pi. But I don't now what exactly caused this error. Just for the case I was incredibly stupid to the resolution to 0×0 by accident in all test cases, I will add an `ArgumentOutOfRangeException` in my next pull request.

In case I experience this problem again, I will reopen this issue and send detailed `vcdgb` information.

Thank You
Daniel

closed 2018-05-19 by daniel-lerch

techyian, 2018-05-19:
Hi Daniel,

Really glad you've been able to get it sorted. Additional handling for this component is a good idea. I must admit I wasn't aware it would behave it that way if the resolution was set to 0x0 but I can understand why. Hopefully it won't crop up again - I'm tempted to try out Avalonia, hadn't heard of it until now.

Thanks,
Ian