techyian/MMALSharp #55: Problems with NuGet

daniel-lerch, 2018-06-18:
Hi Ian,

I have just seen that you have published MMALSharp on NuGet. When I first saw this project on GitHub, the only thing I wanted was to download it on NuGet. This finally becomes possible 👍 .

However I experienced a big problem with MMALSharp 0.4.0.178 installed via NuGet:
StyleCop.Analysers completely breaks the development experience because it gets activated in every project using MMALSharp. For the development of your Library it makes sense but I don't want to write my private code like MMALSharp's rules. Usually there should not even appear a dependency for StyleCop.Analysers in a NuGet release package.

And one stylistic advice: Microsoft recommends using stable NuGet packages. NLog for example could be upgraded to a stable version. And for everything that does not come from Microsoft I would recommend using the latest version for compatibility reasons. Although NuGet packages should always be backwards compatible, there are many developers that want to get rid of old methods and simply delete them. To allow users of MMALSharp to use the same library themselves it should target the latest version.

Daniel

techyian, 2018-06-18:
Hi Daniel,

I've just updated the [NuGet package](https://www.nuget.org/packages/MMALSharp/0.4.1) now and bumped the version to 0.4.1. I think I've managed to exclude the StyleCop dependency so if you could give it a test that would be great. I remember having trouble with it last time I added the StyleCop NuGet package in. I've also updated NLog to the latest version. 

Thanks,
Ian

daniel-lerch, 2018-06-18:
Hi Ian,

now it works perfectly. StyleCop.Analysers does not longer appear when using MMALSharp.

Thanks
Daniel

closed 2018-06-18 by daniel-lerch