techyian/MMALSharp #57: Fliename error with FrameToVideo utility (v0.4.2)

rrskybox, 2018-07-02:
When running the simple FrameToVideo sample code, my application is throwing an exception during the FrameToVideo method where it appears that the delimiter for the jpg extension is getting dropped (see below).
 

[image]

The “.” between the filename and extension appears to be being dropped.  I looked at the VideoUtilities code and on line 48, any trailing “.”s appear to be trimmed correctly, but never reinserted before the extension is appended.

My test code follows.  I’ve modified it slightly from the original sample, but only superficially.

 
- Rick

techyian, 2018-07-03:
Thanks for picking this up. In the process of installing ffmpeg so I can look at this for you.

techyian, 2018-07-03:
The fix for this looks to be working correctly. You can pick it up from source or via MyGet pre-release until I next push to NuGet (finishing a few bits off around the preview renderer before doing so). Let me know if you run into any problems with the fix.

closed 2018-07-03 by techyian

rrskybox, 2018-07-03:
Much obliged.  Your project is a great concept with a superb implementation.  I'm looking forward to putting it to work.  I'll wait until it's up on NuGet to continue my experiments.  - Rick

techyian, 2018-07-03:
Thanks Rick, I hope it can be useful to you.

Ian