techyian/MMALSharp #65: Trying to stream to RTMP server on Raspberry Pi throwing exception Unhandled Exception: System.ComponentModel.Win32Exception: No such file or directory

jesses90, 2018-07-12:
Hello,

I'm trying to stream to an RTMP server (Youtube Live) and am getting the following exception:

Unhandled Exception: System.ComponentModel.Win32Exception: No such file or directory
   at System.Diagnostics.Process.ResolvePath(String filename)
   at System.Diagnostics.Process.StartCore(ProcessStartInfo startInfo)
   at System.Diagnostics.Process.Start()
   at MMALSharp.Handlers.FFmpegCaptureHandler..ctor(String argument)
   at MMALSharp.Handlers.FFmpegCaptureHandler.RTMPStreamer(String streamName, String streamUrl)
   at birdnerd.devices.PiCam.<>c__DisplayClass0_0.<<Run>b__0>d.MoveNext() in <WINDOWS_FILE_PATH_HERE>

Is this at all related to https://brockallen.com/2016/09/24/process-start-for-urls-on-net-core/

techyian, 2018-07-12:
Hi,

So to confirm, is this a .NET Core specific issue? The error at first glimpse sounds like FFmpeg isn't installed, can you confirm it is? Do you receive any errors if you run `ffmpeg` in a console window on your Pi?

techyian, 2018-07-12:
I've just had a quick test on my Pi 3 using .NET Core and the `FFmpegCaptureHandler.RawVideoToAvi` method is working as expected so I would expect the Process call on the `FFmpegCaptureHandler.RTMPStreamer` method to start also. It's likely to be an issue with your FFmpeg install in this case. Let me know if you have any other issues and thanks for trying out MMALSharp.

jesses90, 2018-07-12:
Silly me! Did a fresh install of Raspbian a few days ago and forgot to install ffmpeg. I'm getting a broken pipe exception now but can raise a new ticket if I can't resolve it. Thanks for the help!

techyian, 2018-07-12:
No problem, the RTMP streaming can be a little tricky so just let me know if you need any help.

closed 2018-07-12 by techyian