techyian/MMALSharp #84: Function not implemented. Unable to set camera config.

wimeling, 2018-12-09:
I am having a problem using the library i use it on a raspberry pi 3 model b with .net core 2.1. When i run the all the examples i can see it captures always on picture and then crashes with the following exception: 

Successfully processed 2.81mb. 

Unhandled Exception: MMALSharp.MMALNotImplementedException: Function not implemented. Unable to set camera config.
   at MMALSharp.MMALCallerHelper.MMALCheck(MMAL_STATUS_T status, String message)
   at MMALSharp.Components.MMALCameraComponent.Initialise()
   at MMALSharp.MMALCamera.TakeRawPicture(ICaptureHandler handler)

I use this piece of code from the examples:

public async Task TakeTimelapsePicture()
{                        
    MMALCamera cam = MMALCamera.Instance;

    // This example will take an image every 10 seconds for 4 hours
    using (var imgCaptureHandler = new ImageStreamCaptureHandler("/home/pi/images/", "jpg"))
    {
        var cts = new CancellationTokenSource(TimeSpan.FromHours(4));
        var tl = new Timelapse { Mode = TimelapseMode.Second, CancellationToken = cts.Token, Value = 10 };
        await cam.TakePictureTimelapse(imgCaptureHandler, MMALEncoding.JPEG, MMALEncoding.I420, tl);
    }
    
    // Only call when you no longer require the camera, i.e. on app shutdown.
    cam.Cleanup();
}

closed 2018-12-09 by wimeling

techyian, 2018-12-10:
Hi, what was the cause of the issue here?

wimeling, 2018-12-10:
> Hi, what was the cause of the issue here?

It seemed it had something to do with configuring my raspberry pi because after a reboot it was working as expected. I could not reproduce the issue