techyian/MMALSharp #86: Splitter component video recording

techyian, 2019-01-25:
There appears to be a big issue with video recording when using the splitter component. The resulting video has a rapid playback speed when viewed in VLC. Needs some more investigation as I believe this has regressed for v0.5.0.

techyian, 2019-04-03:
I seem to get the same behaviour when using raspivid also which may indicate a problem with my SD card. Could anyone using MMALSharp please try video recording and let me know if they have similar problems? The video port callback code is completing very quick when doing perf testing (~1ms on average) so I'm struggling to see where the bottleneck may be in MMALSharp itself.

techyian, 2019-04-11:
Tried on another Pi (2B) and different camera module but still seeing same issue. Raised on Pi forums.

inf1981, 2019-05-02:
Hi @techyian,
tried it today and also having the same issue. Using an IMX219 with RPI3. Checked the memory card, it's not related.

techyian, 2019-05-02:
Hi, could you try recording with raspivid and see if you get the same problem? I was seeing the same behaviour with that too which is why I thought the SD card might be the cause.

inf1981, 2019-05-02:
I tried and I think the issue is really the IO speed. So I did this: mount some ramdisk of a size you want to record: _sudo mount -t tmpfs -o size=100M none /home/pi/tmpfs_
Then record a video in there, transfer it and check with VLC by calling in Windows CMD: "C:\Program Files (x86)\VideoLAN\VLC\vlc.exe" --demux=h264 videoname.avi
It worked for me, question is, if there will be some overflow at some point. Increasing the GPU mem to 256M seems a good solution, but I'm not sure yet if it helps. I will investigate more on this and send you updates later!

techyian, 2019-05-02:
Thanks so much. Really appreciate you taking the time to look into it and your findings line up with what I was thinking. Which SD card are you using with your Pi? Looking forward to hearing back from you.

techyian, 2019-06-15:
I'm closing this issue. VLC seems to have some problems playing the containerless format. As a workaround, I converted the video to MKV as follows:

```
sudo apt-get install mkvtoolnix
mkvmerge test.h264 -o test.mkv
```

Where `test.h264` is the name of your video file.

closed 2019-06-15 by techyian