techyian/MMALSharp #90: Fixed/custom name on taken pictures

harniher, 2019-04-05:
If you could have the ability to give the pictures a custom name will be great and useful, also if no name is given, the API saves the photo like the current verion does (date/time format).
Thank you

techyian, 2019-04-05:
Thanks, will try and get this into the v0.5.1 release.

harniher, 2019-04-05:
Thanks to you Ian, for your time and effort MMALSharp is a great API, and
thanks for the fast response.

techyian, 2019-04-07:
Hi @harniher, this work is now in the dev branch if you want to build from source. I have added an additional constructor to the `FileStreamCaptureHandler` class which takes a single string for the full file path. All subclasses of `FileStreamCaptureHandler` also implement this new constructor.

Let me know if you run into any issues with it. A new release will be done soon with this work in.

closed 2019-04-08 by techyian

harniher, 2019-04-08:
Hi @techyian , 
I've tested the code and it works like charm just one little bug that doesn´t affect the performance of the API, when it takes the photo the xterminal shows this

> Preparing to take picture. Resolution: 0 x 0. Encoder: JPEG. Pixel Format: I420.

before the update it showed the actual resolution of the picture taken, the picture saved is of the defined resoulition  (640x480 in this case), just wanted to let you know this bug.
Thanks for the prompt response/update

techyian, 2019-04-08:
Thanks for letting me know :) I'll look into that, it'll have regressed from the work done for #88. Glad to hear it's working for you.

techyian, 2019-04-08:
Just pushed a fix for this so should hopefully be ok now. It was getting the resolution from the Image Encoder which doesn't have a width/height stored and so was just outputting 0.