techyian/MMALSharp #92: Segmentation fault when starting ProcessAsync when remote debugger is attached

KevinBalz, 2019-06-27:
When starting the code example below, and attaching the visual studio debugger over ssh, the program quits after ~2 seconds (due to 'Task.Delay`) with this output
```
Waiting for debugger to attach
Debugger attached
Before ProcessAsync
Segmentation fault
```

```csharp
using System;
using System.Diagnostics;
using System.Threading;
using System.Threading.Tasks;

using MMALSharp;
using MMALSharp.Components;
using MMALSharp.Handlers;
using MMALSharp.Native;
using MMALSharp.Ports;

namespace mmaltest
{
    class Program
    {
        static async Task Main(string[] args)
        {
            Console.WriteLine("Waiting for debugger to attach");
            while (!Debugger.IsAttached) await Task.Delay(10);
            Console.WriteLine("Debugger attached");
            await TakePictureFromVideoPort();
            Console.WriteLine("After Taking Picture");
        }

        public static async Task TakePictureFromVideoPort()
        {
            MMALCamera cam = MMALCamera.Instance;

            using (var imgCaptureHandler = new ImageStreamCaptureHandler("/home/pi/images/", "jpg"))
            using (var splitter = new MMALSplitterComponent(null))
            using (var imgEncoder = new MMALImageEncoder(imgCaptureHandler, continuousCapture: true))
            using (var nullSink = new MMALNullSinkComponent())
            {
                cam.ConfigureCameraSettings();

                var portConfig = new MMALPortConfig(MMALEncoding.JPEG, MMALEncoding.I420, 90);

                // Create our component pipeline.         
                imgEncoder.ConfigureOutputPort(portConfig);

                cam.Camera.VideoPort.ConnectTo(splitter);
                splitter.Outputs[0].ConnectTo(imgEncoder);
                cam.Camera.PreviewPort.ConnectTo(nullSink);

                // Camera warm up time
                await Task.Delay(2000);

                CancellationTokenSource cts = new CancellationTokenSource(TimeSpan.FromSeconds(15));

                // Process images for 15 seconds.        
                Console.WriteLine($"Before {nameof(cam.ProcessAsync)}");
                await cam.ProcessAsync(cam.Camera.VideoPort, cts.Token);
                Console.WriteLine($"After {nameof(cam.ProcessAsync)}");
            }

            // Only call when you no longer require the camera, i.e. on app shutdown.
            cam.Cleanup();
            Console.WriteLine($"Camera cleanup");
        }
    }
}
```

When not attaching a debugger at all or attaching after `ProcessAsync` everything works as expected.

techyian, 2019-06-27:
I've never used the debugger with the library before, and didn't have access to the debugger whilst developing it so I'm not sure why you're receiving a segmentation fault. There is a lot of interop going on with p/invoke so that might be the cause, especially when the native function pointers are called. Are you sure the Raspberry Pi supports debugging via visual studio? I know Windows IoT works fine but I'm not sure Linux support is as refined. 

I don't think this is the fault of the library but more so debugging support on the Pi.

KevinBalz, 2019-06-27:
Debugging seems to work fine, setting breakpoint and stepping works without a problem until it hits `cam.ProcessAsync`. Of course it's possible that there is a bug in the debugging tools itself, considering the library is probably to hit more edge cases through p/invoke etc.