techyian/MMALSharp #95: File to Splitter

KevinBalz, 2019-08-05:
Hi,

is it possible to read a video file similar to the "Encode / Decode from Stream - Video", but instead of outputting to a file, connect it to a splitter/resizer etc. ?

Thanks in advance!

techyian, 2019-08-05:
Hi Kevin,

In theory yes it should be possible, however due to the way the MMALSharp specific File Encoder/Decoder components (e.g `MMALVideoFileEncoder`) are wired up, it makes them unable currently to be connected to other components. I'm in the process of re-working quite a bit of the library for v0.6 so will certainly add this to the backlog for investigation on how to best add this functionality.

techyian, 2019-09-05:
Making progress with this issue. In the 0.6 branch I'm now able to connect an Image Decoder to an Image Encoder and do the conversion as a single operation which is great. Having some issues currently with Video Decode - Video Encode where it appears to hang half way through processing...will keep playing around with this to get to the bottom of it.

techyian, 2019-09-05:
After further investigation I've found some inconsistencies between Pi 3b+ (Stretch) and Pi 4 (Buster). This is what I have found so far:

**Pi 4**

Test 1
-------
Video Decoder (H.264 / I420) single operation
Video Encoder (H.264 / I420) single operation

Results in a file created that only shows essentially the 1st frame of the video file used.

Test 2
-------
Video Decoder (H.264 / I420) -> Video Encoder (H.264 / I420) connected together will hang part way through operation. Strangely enough, re-encoding to MJPEG will complete the operation, resulting in a file containing the 1st frame of the video.

Throwing a splitter into the mix doesn't fix it either.

**Pi 3**

Test 1
-------
Same issue as Pi 4.

Test 2
-------
This works perfectly, the re-encoded video is nearly identical to the original. I have also tried connecting a splitter and this works too.

N.B I feel there may be something wrong with the decoder component. Using a 3.5mb H.264 video, the resulting raw decoded file is only 8mb which I think is too low. Need to investigate this further as that might explain why. Will regression test against v0.5.1 as single operations to see what the outputted raw file size is to compare.

techyian, 2019-09-06:
Confirmed the Video Decoder has regressed in v0.6 which explains the issue with the resulting video only showing the first frame. Will investigate.

techyian, 2020-01-02:
All functionality appears to be working well for this ticket. I've added examples to the Wiki, see [here](https://github.com/techyian/MMALSharp/wiki/Advanced-Examples#Encode-Decode-Video-FileStream-Decode2Splitter2Encode) and [here](https://github.com/techyian/MMALSharp/wiki/Advanced-Examples#Encode-Decode-Video-FileStream-Decode2Splitter2Resizer2Encode)

closed 2020-01-02 by techyian