techyian/MMALSharp #101: Timelapse Resolution 0x0.

adamstenbaek, 2019-11-20:
Hello!
I'm trying to make a time-lapse with this. However it wont work. 
All i get is "Preparing to take picture. Resolution: 0 x 0. Encoder: JPEG. Pixel Format: I420"

techyian, 2019-11-20:
Hey,

Can you confirm which version you're using please? Have you cloned from the "dev" branch? If so, that's still in alpha and I'm fully aware there's some outstanding issues which need fixing. If you're using v0.5.1 from NuGet and having issues then I'll investigate for you although the timelapse unit tests were passing on that version.

Thanks.

adamstenbaek, 2019-11-20:
Im using the Nuget v5.0.1.
I just tried using  video. And it seems fine

adamstenbaek, 2019-11-20:
I also dont know much about how this works.
Anything i need to do?

techyian, 2019-11-20:
Are you following the example in the [Wiki](https://github.com/techyian/MMALSharp/wiki/Examples#timelapse-mode)? Also, which runtime are you using, Mono or .NET Core?

adamstenbaek, 2019-11-20:
Yes. I am using the example on the wiki with a slight difference: I'm using FromDays.
And I'm using .NET Core

techyian, 2019-11-20:
Yes it looks as though there is a bug here. I will raise it and fix ASAP. Thank you for letting me know. It'll likely be patched into the v0.6 release.

closed 2019-11-21 by adamstenbaek