techyian/MMALSharp #105: (Community) Feature requests

techyian, 2019-12-30:
Hello MMALSharp users!

I'm raising this ticket to ask if there are any specific features you would like me to try and implement in the library for upcoming releases. Please don't be shy!

MV10, 2020-07-05:
Configurable format strings for date/time overlays would be nice.

MV10, 2020-07-08:
How about the option to load a motion-detection mask? A simple black & white image would suffice, just logical-AND with the frames being compared and the black areas will always be identical. I think you keep the previous new frame for the next comparison right? So that would be just one logical-AND per comparison, I'd think that's minimal overhead.

Can you easily read a standard-format graphics file and convert it to whatever raw format is used for motion-detection, or would the user have to supply one in a suitable raw format? That could be a problem I suppose.

MV10, 2020-08-10:
> Configurable format strings for date/time overlays would be nice.
> Hmm, I'm surprised the date/time isn't ever updated...?

Quoting myself... 😁 

I did some poking around, and I see how the date/time overlay is generated. It turns out if I set up a little task which is mostly idle but calls `cam.EnableAnnotation` when the time changes, this is enough to generate an overlay with the updated time, and it doesn't appear to cause any problems. That call runs very quickly (0.4ms to 0.5ms on average) so it would be a nice-to-have, perhaps as part of `ProcessAsync` when `ShowTimeText` is enabled.

I'd be happy to PR the change. Would you also be open to config options to specify the date and time formats?