techyian/MMALSharp #125: Instance access - liblibbcm_host.so: cannot open shared object file: No such file or directory

thnk2wn, 2020-02-21:
* MMALSharp 0.5.1
* 3.1.1-buster-slim-arm32v7 docker image (running privileged)
* OS Raspbian GNU/Linux 10 (buster)
* Raspberry Pi 4 Model B Rev 1.1 (4GB RAM)

When I reference `MMALCamera.Instance` I get the following exception.

<pre>
Unhandled exception. System.DllNotFoundException: Unable to load shared library 'libbcm_host.so' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibbcm_host.so: cannot open shared object file: No such file or directory
   at MMALSharp.Native.BcmHost.bcm_host_init()
   at MMALSharp.MMALCamera..ctor()
   at MMALSharp.MMALCamera.<>c.<.cctor>b__28_0()
   at System.Lazy`1.ViaFactory(LazyThreadSafetyMode mode)
   at System.Lazy`1.ExecutionAndPublication(LazyHelper executionAndPublication, Boolean useDefaultConstructor)
   at System.Lazy`1.CreateValue()
   at System.Lazy`1.get_Value()
   at MMALSharp.MMALCamera.get_Instance()
</pre>

The camera is enabled and I was able to test it via `raspistill -o image.jpg`. Is there something else I have to install or configure on the device to use this? Ideas?

techyian, 2020-02-21:
I'm not overly familiar with docker and have never tried running the library in a container, but does the following help? https://github.com/skonagaya/RPi-Streamer/blob/master/Dockerfile#L75. Symlinks are being made to `/usr/lib/` which I suspect is so that application can view the relevant libraries needed. Unsure whether you'll need all of them but it might put you in the right direction?

thnk2wn, 2020-02-21:
Thanks, that helps. I also see others with Dockerfile steps like:

````
RUN apt-get install libraspberrypi-bin -y
RUN usermod -a -G video root
````

I'll do some research, just wasn't sure what all dependencies this library had in terms of file access, permissions, etc. that might need to be accounted for when running containerized.

techyian, 2020-02-21:
Off the top of my head, I think it's `libbcm_host.so` and `libmmal.so` the library needs access to. Let me know how you get on. On a Raspbian based OS `libraspberrypi-bin` should already be installed but you may need that step if running on Ubuntu for example.

thnk2wn, 2020-02-24:
I added `-v /opt/vc/lib:/opt/vc/lib` into my docker run command and now the original error went away.

Now I have a similar one but for `libmmal.so`. Where should that be?

<pre>
System.DllNotFoundException: Unable to load shared library 'libmmal.so' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibmmal.so: cannot open shared object file: No such file or directory
   at MMALSharp.Native.MMALComponent.mmal_component_create(String name, IntPtr* comp)
   at MMALSharp.MMALComponentBase.CreateComponent(String name)
   at MMALSharp.MMALComponentBase..ctor(String name)
   at MMALSharp.Components.MMALCameraComponent..ctor()
   at MMALSharp.MMALCamera..ctor()
   at MMALSharp.MMALCamera.<>c.<.cctor>b__28_0()
   at System.Lazy`1.ViaFactory(LazyThreadSafetyMode mode)
   at System.Lazy`1.ExecutionAndPublication(LazyHelper executionAndPublication, Boolean useDefaultConstructor)
   at System.Lazy`1.CreateValue()
   at System.Lazy`1.get_Value()
   at MMALSharp.MMALCamera.get_Instance()
</pre>

I also set `LD_DEBUG=all` and can dump that if helpfu.

thnk2wn, 2020-02-24:
I got it working. I had to add `ENV LD_LIBRARY_PATH /opt/vc/lib` to my Dockerfile and I added some additional volume mounts. However I'm not sure some of the volume bindings I tried along the way were strictly required so I may try removing so I only end up with what's actually needed.

It did appear to silently fail when capturing the image and specifying a path that didn't exist but I didn't have the destination (`/home/pi/motion-media`). mounted at that point.

```bash
docker run \
    --privileged \
    -v /opt/vc/lib:/opt/vc/lib \
    -v /usr/lib/arm-linux-gnueabihf:/usr/lib/arm-linux-gnueabihf \
    -v /lib:/lib \
    -v /home/pi/motion-media:/home/pi/motion-media \
    -d \
    --name $name \
    $username/$name:latest
````

closed 2020-02-24 by thnk2wn

techyian, 2020-02-24:
Excellent, glad it's working for you.