techyian/MMALSharp #126: System.DllNotFoundException

ggggg15, 2020-02-23:
When 
MMALCamera cam = MMALCamera.Instance; 
is executed I get the following error:
System.DllNotFoundException: 'Unable to load DLL 'libbcm_host.so' or one of its dependencies: The specified module could not be found. 

Raspberry PI model 3B
Windows 10 IoT version 1809 (build 17763) 
Visual Studio 2019 C#
MMALSharp v0.51

techyian, 2020-02-23:
This library is only compatible with Linux based operating systems which include the Raspberry Pi camera firmware. Windows 10 IoT does not have the camera module firmware available unfortunately. If you want to use this lib, please use it under an OS such as Rasbian or Ubuntu.

ggggg15, 2020-02-23:
Ok That explains it all. Thanks.

closed 2020-02-23 by ggggg15