techyian/MMALSharp #127: NLog Dependency Exists on NuGet, conflicts with README?

thnk2wn, 2020-02-25:
The README indicates that v0.6 uses Microsoft Abstraction Logging but the latest version on NuGet (https://www.nuget.org/packages/MMALSharp/) is 0.5.1 and that has a large NLog dependency I'd like to remove.

I might be missing something but is there a version missing on NuGet or is that prerelease elsewhere or maybe the README is out of date?

techyian, 2020-02-25:
You can pick up the prerelease binaries from [MyGet](https://www.myget.org/feed/mmalsharp/package/nuget/MMALSharp) which contains the `Microsoft.Extensions.Logging.Abstractions` work. Alternatively, you can clone from the dev branch. The examples in the Wiki are up to date for v0.6. Working on a few final tickets before an official release of 0.6 can be done.

thnk2wn, 2020-02-26:
Thanks, that worked

closed 2020-02-26 by thnk2wn