techyian/MMALSharp #129: TakePictureTimeout. Argument is invalid. Unable to set camera config. Buffer null, could not release

thnk2wn, 2020-02-28:
One more thing I'm noticing while testing. Same code, setup and version as #128. I see this error and then see the warning logged "Buffer null, could not release" every second without stop. 

<pre>
[19:49:49 INF] Successfully processed 0.00mb.
[19:49:49 ERR] Error taking picture
MMALSharp.MMALInvalidException: Argument is invalid. Unable to set camera config.
   at MMALSharp.MMALNativeExceptionHelper.MMALCheck(MMAL_STATUS_T status, String message)
   at MMALSharp.MMALCameraComponentExtensions.SetCameraConfig(MMALCameraComponent camera, MMAL_PARAMETER_CAMERA_CONFIG_T value)
   at MMALSharp.Components.MMALCameraComponent.Initialise(IOutputCaptureHandler stillCaptureHandler, IOutputCaptureHandler videoCaptureHandler)
   at MMALSharp.MMALCamera.ConfigureCameraSettings(IOutputCaptureHandler stillCaptureHandler, IOutputCaptureHandler videoCaptureHandler)
   at MMALSharp.MMALCamera.TakePictureTimeout(IFileStreamCaptureHandler handler, MMALEncoding encodingType, MMALEncoding pixelFormat, CancellationToken cancellationToken, Boolean burstMode)
   at CatSiren.Siren.MotionSensorHostedService.CaptureFootageAsync() in /build/MotionSensorHostedService.cs:line 108
   at CatSiren.Siren.MotionSensorHostedService.MotionStateChanged(Object sender, MotionState state) in /build/MotionSensorHostedService.cs:line 89
[19:49:50 WRN] Buffer null, could not release.
[19:49:51 WRN] Buffer null, could not release.
[19:49:52 WRN] Buffer null, could not release.
[19:49:53 WRN] Buffer null, could not release.
[19:49:54 WRN] Buffer null, could not release.
</pre>

techyian, 2020-02-28:
That issue normally occurs either when:

a) A user attempts to use the camera after the Broadcom library has been unloaded.
b) Resources haven't been released cleanly, either due to an issue with the user's code, or the application has been forced to close.

A reboot would usually fix this, but if it's happening reliably in your application then there's something going wrong somewhere.

I believe the `AddHostedService` method in your Startup class adds the background service as a Singleton object which is correct for what MMALSharp needs (only one instance of the camera can ever be created at a given time). Does this error start happening after your Dispose method is called in `MotionSensorHostedService`? I wonder whether there is an issue bootstrapping the Broadcom library after it's been unloaded?

thnk2wn, 2020-02-29:
Thanks again @techyian. Yeah it appeared to happen only after stopping the app then starting it again. This appeared to be this issue: https://github.com/dotnet/extensions/issues/1363 (Hang when calling start/stop on IHost). There was a workaround at the end that appeared to allow proper shutdown. So far I can't reproduce it since.

closed 2020-02-29 by thnk2wn

techyian, 2020-02-29:
Excellent, glad you got it fixed!