techyian/MMALSharp #137: AWB Mode "Greyworld" missing

Spontifixus, 2020-04-10:
I am using the NoIR version of the camera. To get a decent colored image I use the AWB-mode "greyworld", as defined in [raspistill documentation](https://www.raspberrypi.org/documentation/raspbian/applications/camera.md).

The enum `MMAL_PARAM_AWBMODE_T` defined in [MMALParameters.cs](https://github.com/techyian/MMALSharp/blob/dev/src/MMALSharp/Native/MMALParameters.cs) lacks the required member, but unfortunately I have no idea where to find the values to be able to fix this myself and create a pull request.

Can you point me in the right direction where to find the relevant docs?

techyian, 2020-04-11:
This is related to #111 and I have linked to the commit in the userland repository where the member was added.

Spontifixus, 2020-04-11:
Note so self: Do not open issues that late in the night. Thank you for clarifying.

closed 2020-04-11 by Spontifixus

techyian, 2020-04-11:
AWB Grey World option added and pushed to dev branch.