techyian/MMALSharp #149: Crop-Rect does not have an effect

CobraCalle, 2020-05-18:
Hello,

I would like to crop the image from the camera to a specific rect, but this does not have any effect... 

Here is my code:
            MMALCamera cam = MMALCamera.Instance;

            MMALCameraConfig.Resolution = Resolution.As2MPixel;
            MMALCameraConfig.EncodingSubFormat = MMALEncoding.BGR24;
            MMALCameraConfig.VideoStabilisation = false;

            var cropRect = new System.Drawing.Rectangle(177, 64, 1174, 1136);

            using (var imgCaptureHandler = new OpenCvMatCaptureHandler(MMALCameraConfig.Resolution))
            using (var splitter = new MMALSplitterComponent())
            using (var nullSink = new MMALNullSinkComponent())
            {
                cam.ConfigureCameraSettings();

                var splitterInputConfig = new MMALPortConfig(MMALEncoding.OPAQUE, MMALEncoding.BGR24, crop: cropRect);
                var splitterOutputConfig = new MMALPortConfig(MMALEncoding.BGR24, MMALEncoding.BGR24);

                // Create our component pipeline.        
                splitter
                    .ConfigureInputPort(splitterInputConfig, cam.Camera.VideoPort, null)
                    .ConfigureOutputPort<FastStillPort>(0, splitterOutputConfig, imgCaptureHandler);

                cam.Camera.VideoPort.ConnectTo(splitter);
                cam.Camera.PreviewPort.ConnectTo(nullSink);

                // Camera warm up time
                await Task.Delay(2000);

                CancellationTokenSource cts = new CancellationTokenSource(TimeSpan.FromSeconds(10));

                await cam.ProcessAsync(cam.Camera.VideoPort, cts.Token);
            }

techyian, 2020-05-18:
I believe you will want to set the ROI against the camera itself and you should be using `MMALCameraConfig.ROI` which accepts a Zoom struct. Valid range is between 0 - 1.0.

CobraCalle, 2020-05-18:
yes, I treid that... but if I apply the values from the wiki (0.5, 0.5, 0.1, 0.1) the resulting frames are still 1600x1200 (in my case)... as I understand those values the result should be 10% of the height and width from the center... or is it allways a zoom and it is not possible to crop an area to reduce the frame size (and speed up the fps)?

techyian, 2020-05-18:
It isn't a physical "crop" as you would usually expect, but instead it will focus on an area of the image keeping the original resolution. That's how the native MMAL library works so I don't have any control over that unfortunately.

CobraCalle, 2020-05-18:
OK... checked.. thanks a lot 👍

closed 2020-05-18 by CobraCalle