techyian/MMALSharp #150: corrupted size vs. prev_size

CobraCalle, 2020-05-25:
Hello,

when I take a still image with the following code, it works one time (but takes mutliple secods to respond), And after the first try, I always get the following error and the process is terminated...

corrupted size vs. prev_size

Does anyone know what does this mean?

techyian, 2020-05-25:
Hi,

You haven't provided a code sample in your message, please can you provide one so I can see what you're trying to do? That doesn't look like an error I've come across before so it's likely an issue with your code.

CobraCalle, 2020-05-25:
This is how I`ve initialized the cam:

            var cam = MMALCamera.Instance;

            MMALCameraConfig.Resolution = Resolution.As2MPixel;
            MMALCameraConfig.EncodingSubFormat = MMALEncoding.BGR24;

            cam.ConfigureCameraSettings();

And this is how I`ve taken the picture (first time after reboot works, after that I get an error):
            using (var imgCaptureHandler = new MemoryStreamCaptureHandler())
            {
                await this.camera.TakePicture(imgCaptureHandler, MMALEncoding.JPEG, MMALEncoding.I420);

                return imgCaptureHandler.CurrentStream.ToArray();
            }

CobraCalle, 2020-05-25:
additional information... the error occurs even if I do not take another picture... just wait after taking the first picture

techyian, 2020-05-25:
I'm really not sure how you've managed to get that error. I can't reproduce this when testing locally with that code sample. Are you doing anything fancy with the buffer headers as we've discussed in #146 or is this code in any way related to what you're trying to achieve in that issue?

CobraCalle, 2020-05-25:
No... Nothing special here. My application allows the user to grab a frame from the camera in the settings of the camera. So that was the first thing I implemented for the mmal based camera. It's just that... Could it be related to the hq camera? Is there another way to grab a single frame?

techyian, 2020-05-26:
> Could it be related to the hq camera?

Perhaps, but as I don't have one of those camera modules it's difficult to say. Have you got the latest firmware installed? 

I noticed that you're sending BGR24 from the camera (`MMALCameraConfig.EncodingSubFormat = MMALEncoding.BGR24;`) but then doing a format conversion on the `await this.camera.TakePicture(imgCaptureHandler, MMALEncoding.JPEG, MMALEncoding.I420);` call back to YUV420 - was this deliberate? In addition, the `TakePicture` helper method you're calling implicitly calls `ConfigureCameraSettings` on your behalf so you're not required to call it yourself, please see [here](https://github.com/techyian/MMALSharp/blob/dev/src/MMALSharp/MMALCamera.cs#L199).

CobraCalle, 2020-05-26:
I'll close this issue, because I don't think it is realted to MMAL... this happens when using MMAL, but I think it is realted to a corrupted file system, or the power supply, I`m using at the moment.

closed 2020-05-26 by CobraCalle

techyian, 2020-05-26:
Ok no problem. If you're ever unsure about a hardware issue then using raspistill is a good test to carry out.