techyian/MMALSharp #152: Faster image capture

jonmiller1, 2020-06-14:
I'm using the V2.1 camera module.  I'm trying to get full resolution images out as fast as possible.  I've tried using the video port but the images are too blurry.  I need to be able to read a barcode out of the image.  The full frame works but it is a bit slow.  What kind of optimizations can I make?  Can I make the video port sharper?  I only need a gray scale image, is there anyway to instruct the camera to only return grayscale data?  Would that be faster?  Any suggestions would be appreciated.  This library is awsome.  You've done good work here.

techyian, 2020-06-14:
Hi Jon,

Thanks for your feedback :) As you've noticed with the video port, although it can process frames much quicker there is a tradeoff in that it doesn't implement the aggressive denoise algorithm that the still port does. Have you tried enabling video stabilisation with `MMALCameraConfig.VideoStabilisation`? Please be aware though that when enabled, this feature will crop a small region of your image frames to compensate for reducing artifacts. I'm unsure whether this will improve things to a standard good enough for your application but unfortunately this library is limited to what's offered by the native MMAL library and camera hardware. 

> I only need a gray scale image, is there anyway to instruct the camera to only return grayscale data?

There isn't a native way to output greyscale from the camera itself, this would be handled by stripping the luma component of a YUV420 frame, but that means addition work on your part and wouldn't improve performance; the speed to store frames once they're outputted from the camera itself is pretty quick.

> Can I make the video port sharper?

You could try playing around with the various configuration settings in the `MMALCameraConfig` class, specifically `MMALCameraConfig.Sharpness`.

Other than that I'm not sure what more you can do to improve the video port's quality for image stills unfortunately.

Ian

techyian, 2020-06-14:
My comments around enabling video stabilisation only apply if you're cloning the latest code from this repo. The official NuGet release has this enabled by default.

techyian, 2020-07-17:
Closing. Please re-open if you're still having issues.

closed 2020-07-17 by techyian