techyian/MMALSharp #155: Logging debug output with Serilog

MV10, 2020-07-07:
Here's a note for your [Logging](https://github.com/techyian/MMALSharp/wiki/Configuration#logging) topic in your wiki about how to set up Serilog. It doesn't normally work with `LoggerFactory` the way older loggers do (I used to be a frequent Serilog contributor), so it's kind of hard to track down how this is done because one of the packages has been mostly superseded by builder-helpers in other packages. The dependencies are _Microsoft.Extensions.Logging_, _Serilog.Extensions.Logging_, and _Serilog.Sinks.File_.

There are many other possible approaches, but at a minimum this gets the job done:

```csharp
if (useDebug)
{
    File.Delete("debug.log"); // log output will append
    MMALCameraConfig.Debug = true;
    MMALLog.LoggerFactory = new LoggerFactory()
        .AddSerilog(new LoggerConfiguration()
            .MinimumLevel.Verbose()
            .WriteTo.File("debug.log")
            .CreateLogger());
}
```

techyian, 2020-07-07:
Hi Jon,

Thanks for the info on setting up Serilog, I'll get that added to the Wiki. I've been trying to keep up with the other issues you've raised and happy to review any changes you feel would be appropriate when piping to StdOut/Err. I've not done much research into using the new `Channel` APIs, but the problem I have with using async code in that section of the library is that currently you're unable to `await` anything from the native callback methods as they're unsafe due to accepting native pointers as parameters; this could potentially be changed and accept `IntPtr` types instead, but I'd need to be clear on the performance benefits before making such changes as it'll get messy quick with casts all over the place. If you want to send a PR in with your proposed changes I'll review them and we can take it from there.

Thanks,
Ian

MV10, 2020-07-07:
Yeah I was going in about three directions at once in that other topic, sorry about that. The MMALCamera.Cleanup exception I mentioned over there was because I was using the new one-liner `using` statement (no block in braces) which meant nothing was disposed until the method exits -- and I was calling Cleanup before that. So no problems there but my own.

Since you already had `ProcessAsync` running the show, the additional async changes didn't take over the world after all. Before I PR anything maybe have a peek at what `ExternalProcessCaptureHandler` turned into ([my repo](https://github.com/MV10/pi-cam-test)). It's considerably improved over what I started with the other day. If you like it, I'll PR it into this project. In addition to buffering the output, I also added the ability to specify one or more signals to cleanly terminate signal-aware apps. While that doesn't help with ffmpeg due to its quirks I mentioned in the other thread, it does allow clean termination of VLC, which I'm using for preview-quality MJPEG streaming direct to browsers.

From a package-consumer standpoint there are just two changes:

* creating the handler (all but file & args have defaults) [(link)](https://github.com/MV10/pi-cam-test/blob/master/Program.cs#L169)
* await another task along with `ProcessAsync` [(link)](https://github.com/MV10/pi-cam-test/blob/master/Program.cs#L192)

That extra task manages buffering and shutdown:

* internally starts another task to manage the buffer output
* waits for the token to cancel (signals the end of `ProcessAsync`)
* leaves the output buffer running while raising any shutdown signals
* short delay to let the process exit and flush the buffer, then ends that task

Apart from that it's basically the ffmpeg handler.

techyian, 2020-07-17:
Notes added to wiki.

closed 2020-07-17 by techyian