techyian/MMALSharp #173: Pipeline-based video manipulation

MV10, 2020-08-27:
Is it currently possible to tap into a video capture handler (or descendant such as `CircularBufferCaptureHandler`) to post-process frame data? I see it exposes `Manipulate` but I assume that's intended for still-frame processing (and I noticed your performance note in #108 while searching about this topic).

As you know, I'm doing a lot of work related to motion detection, and it would be nice to output video which is modified to visualize the different techniques I'm using -- either live from the camera, or from file input (to create reproducible tests). It would also be useful to end-users for tuning the frame-diff threshold and any other parameters which may be introduced later.

I know how to do this with internal changes to MMALSharp, but I can't seem to work out how to do it "from the outside" via the pipeline -- but honestly I still find the pipeline relationships a little confusing.

techyian, 2020-08-27:
Yes, the `Manipulate` method at present is intended for a full still image processing and is called via the `PostProcess` method which in turn is called when the `ProcessAsync` method is finishing. If you want to do things per-frame, I've been telling users to implement their own capture handler and add a delegate they can hook onto from elsewhere. I maybe get the impression you don't like implementing your own custom capture handlers and would prefer there to be features such as this baked into the library? I suppose we could start a discussion around this as it might make things easier? Open to suggestions.

MV10, 2020-08-27:
I just wanted to confirm I hadn't overlooked something. I'll play around with a custom capture handler and circle back to this if it starts feeling like something that could be usefully generalized across handlers. Thanks!

closed 2020-08-27 by MV10