techyian/MMALSharp #191: Setting ShutterSpeed not taking

stwolfe, 2020-10-19:
I have tried simple snapshot using the settings you have in the example docs:

MMALCameraConfig.ShutterSpeed = 2000000; // Set to 2s exposure time. Default is 0 (auto).
MMALCameraConfig.ISO = 400; // Set ISO to 400. Default is 0 (auto).

When inspecting the image that is taken, the ISO is correct but the shutter speed used is always 1/30s.

stwolfe, 2020-10-19:
I see the settings for framerate have changed in the dev branch from master.  And I have tried to compare what is going on in MMALSharp vs Raspistill.

Raspistill setting just the ShutterSpeed does works.

techyian, 2020-10-20:
The shutter speed is constrained by the camera still port's frame rate. Raspistill uses a variable frame rate by setting the value to 0. If you want to see the same behaviour as Raspistill, depending on whether you're using v0.6 or the current dev branch, you'll want to do one of the following:

For v0.6 (current NuGet build): `MMALCameraConfig.VideoFramerate = new MMAL_RATIONAL_T(0, 1);` - see [here](https://github.com/techyian/MMALSharp/issues/138#issuecomment-613428104) for context.

For v0.7 (current dev branch): `MMALCameraConfig.Framerate = 0;`

Please let me know if that fixes the problem for you.

stwolfe, 2020-10-20:
@techyian,

Yes, setting the VideoFramerate allowed the shutter speed to take.   I was setting the StillFramerate, I assume that bit of confusion is why the dev build only has 1 setting.

The only difference I see between my simple application just setting the VideoFramerate and shutter speed is the selected resolution.   My program with MMALSharp takes picture @ 2560 x 1920, where as rapsistill with just setting shutter speed takes picture @ 3280 x 2464

Thanks for the quick response, this has me moving forward.

techyian, 2020-10-20:
No problem. On v0.6, `MMALCameraConfig.StillResolution` is set to `Resolution.As5MPixel` by default and you can override it with your desired resolution.

stwolfe, 2020-10-20:
Thanks again for your responses, and thanks for creating such a great library.

closed 2020-10-20 by stwolfe