techyian/MMALSharp #195: Default motion detection example for v0.6 throws "Unhandled exception. System.TypeInitializationException: The type initializer for 'Gdip' threw an exception."

IKoshelev, 2021-02-14:
I have a .NET 5 app, referencing MMALSharp v0.6, building and running on Pi 4 (2Gb) Raspbian Lite, copypasted default motion detection example verbatim from https://github.com/techyian/MMALSharp/wiki/Advanced-Examples/93b717ebbf4502f3a6c1ef99137a6b416dd0c3e6#FrameDiffDetection , getting an exception:

```
Unhandled exception. System.TypeInitializationException: The type initializer for 'Gdip' threw an exception.
 ---> System.DllNotFoundException: Unable to load shared library 'libgdiplus' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibgdiplus: cannot open shared object file: No such file or directory
   at System.Drawing.SafeNativeMethods.Gdip.GdiplusStartup(IntPtr& token, StartupInput& input, StartupOutput& output)
   at System.Drawing.SafeNativeMethods.Gdip..cctor()
   --- End of inner exception stack trace ---
   at System.Drawing.SafeNativeMethods.Gdip.GdipCreateBitmapFromScan0(Int32 width, Int32 height, Int32 stride, Int32 format, HandleRef scan0, IntPtr& bitmap)
   at System.Drawing.Bitmap..ctor(Int32 width, Int32 height, PixelFormat format)
   at MMALSharp.Processors.Motion.FrameDiffAnalyser.LoadBitmap(MemoryStream stream)
   at MMALSharp.Processors.Motion.FrameDiffAnalyser.PrepareDifferenceImage(ImageContext context, Int32 threshold)
   at MMALSharp.Processors.Motion.FrameDiffAnalyser.CheckForChanges(Action onDetect)
   at MMALSharp.Processors.Motion.FrameDiffAnalyser.Apply(ImageContext context)
   at MMALSharp.Handlers.CircularBufferCaptureHandler.Process(ImageContext context)
   at MMALSharp.Callbacks.PortCallbackHandler`2.Callback(IBuffer buffer)
   at MMALSharp.Callbacks.VideoOutputCallbackHandler.Callback(IBuffer buffer)
   at MMALSharp.Ports.Outputs.VideoPort.NativeOutputPortCallback(MMAL_PORT_T* port, MMAL_BUFFER_HEADER_T* buffer)
```

IKoshelev, 2021-02-14:
Update, issue looks to be fixed by 
```
sudo apt install libc6-dev 
sudo apt install libgdiplus
```
as suggested here https://github.com/dotnet/core/issues/2746

techyian, 2021-02-14:
Yes, I was going to suggest that, apologies - I thought that was already mentioned in the wiki but possibly not. I would definitely advise cloning the master branch as it features many improvements to the motion detection API, including huge performance improvements.

IKoshelev, 2021-02-14:
@techyian Yes, thank you, I cloned and built master on a Win 10 machine, then dropped dlls into project on my Pi, but encountered some strange behavior. When I tried to run a basic video stream,  video was in 1280x720 resolution and trying to change it via `MMALCameraConfig.Resolution = new Resolution(1920, 1080);` lead to an exception

```
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
Unhandled exception. MMALSharp.MMALNoSpaceException: Out of resources. Unable to enable component
   at MMALSharp.MMALNativeExceptionHelper.MMALCheck(MMAL_STATUS_T status, String message)
   at MMALSharp.MMALComponentBase.EnableComponent()
   at MMALSharp.Components.MMALCameraComponent.Initialise(IOutputCaptureHandler stillCaptureHandler, IOutputCaptureHandler videoCaptureHandler)
   at MMALSharp.MMALCamera.ConfigureCameraSettings(IOutputCaptureHandler stillCaptureHandler, IOutputCaptureHandler videoCaptureHandler)
   at StreamSplitterExperiment.Program.CaptureDifferentStreams(OutputWriterHandlerPipeWriter h264PipeWriter, OutputWriterHandlerPipeWriter mjpegPipeWriter, Symaphore symaphore, CancellationToken token) in /home/pi/raseye-sharp/StreamSplitterExperiment/Program.cs:line 121
   at StreamSplitterExperiment.Program.Main(String[] args) in /home/pi/raseye-sharp/StreamSplitterExperiment/Program.cs:line 106
   at StreamSplitterExperiment.Program.<Main>(String[] args)
```

so, I thought Alpha was still at an early stage and went back to V0.6.

P.S. thank you for great lib.

techyian, 2021-02-14:
Thanks for the positive feedback. Could you provide an example code snippet? I may be able to see where the issue is.

IKoshelev, 2021-02-14:
@techyian I sent out an invitation to the private repo with my Pi surveillance fun-project. Link to particular snapshot: https://github.com/IKoshelev/reseye-sharp/blob/1720383726bf936b6fad17b7f3a94a06aeef9bbd/StreamSplitterExperiment/Program.cs 
Mind you, this is just a playground repo, so a lot of code is throw-away.
I run this project on my PI and connect to stream via vlc with 
```
tcp/h264://{my pi IP}:9010
```

IKoshelev, 2021-02-16:
@techyian I've forked this repo, added a playground project and replicated `mmal: mmal_vc_component_enable: failed to enable component: ENOSPC` in it: https://github.com/IKoshelev/MMALSharp/blob/dev/Playground/Program.cs

IKoshelev, 2021-02-27:
@techyian Can you maybe activate discussions for this repo? https://docs.github.com/en/discussions/quickstart

I've done some experiments with 
`System.Numerics.Vector` https://instil.co/blog/parallelism-on-a-single-core-simd-with-c/ 
`System.Runtime.Intrinsics` https://devblogs.microsoft.com/dotnet/hardware-intrinsics-in-net-core/
and was wondering, if we they can improve frame-diffing speed for motion detection, but they require .NET5 and ARM64 (so, 64bit OS).

MV10, 2021-03-12:
@IKoshelev I just ran a quick test and I see that `out of resources` error, too. I'm not quite ready to get active on this project again, but very close. I hope all my motion detection changes didn't introduce a regression!

Because motion detection is my focus, I have been very interested in seeing what the overall .NET 5 improvements look like, but I like your thinking about intrinsics, too. That would be a big change for me (I'm running 32 bit at the moment) but probably worthwhile.

Hi @techyian -- hope you're well.

IKoshelev, 2021-03-14:
@MV10 I tried intrinsics with Raspbian 64. They worked great, the loop that took 60 ms in plain C# took nole 2 ms with intrinsics. BUT Raspbian 64 is in beta and camera does not work yet at all. Not to mention, I'm guessing MMAL bindings for x64 would be different?

CobraCalle, 2021-12-16:
@IKoshelev Any news about the "our of resources" error? I`m using the hq-camera with my pi, but when I change the resolution to more the 2mp I get the "our of resources" error... Would be grate to be able to use the last version with hq-camera...